GenPi64/Build.Dist #1: Documentation

samip5, 2020-12-25:
- [ ] List EVERY FREAKING requirement
- [ ] Make sure ALL scripts use python3.9 and do not try to use python3 binary

perkinslr, 2020-12-27:
I have a more complete listing in the alpha4 branch.  I believe it's everything explicitly needed on gentoo, and what I know is needed elsewhere (but there's probably more if you're on a barebones distro).  When time allows, I'll fire up a `debootstrap` chroot and give it a whirl.

samip5, 2021-02-03:
> I have a more complete listing in the alpha4 branch. I believe it's everything explicitly needed on gentoo, and what I know is needed elsewhere (but there's probably more if you're on a barebones distro). When time allows, I'll fire up a `debootstrap` chroot and give it a whirl.

Have you tried to do that?