GenPi64/Build.Dist #3: json2yaml  (or yaml back to json) for making custom tweaked configs

necrose99, 2020-12-29:
_json2yaml Convert JSON to YAML or vice versa, while preserving the order of associative arrays._  

makes for easier to read , hand hacking ,   also back to json .. for use..     maxs for for easier to ready and or hand edits etc. 
**esp dyslexia**    and bit easier on eyes ..   
check-yaml-config  pass , convert back to json.      
https://www.json2pretty.com/  , https://www.json2yaml.com/  while it adds white spaces... it also makes for easier to read. 

yaml also support comments 
` ### need to do symlink as  temp work around/s from gentoo-overlay-profile ### `

GenPi64json  > add bin-overlay  skell-crossdev.sh (do crossdev-setup.sh  )   (bin-overlay-repo akin to ppa/copr) 
sys-devel/binutils multitarget (xen/crossdev tends to succeed app-emulation/xen ~arm64  may not but crossdev tends to not fubar as often with it )    
install no ..  
bin-overlay=true 
publish-bin-overlay=true 
stage-pack=false
(limit package trees to a few packages ie <crossdev-binhost.genpi.com > needs gentoo/genpi64-minimal  base to run) 
Crossdev priority =1  
`(-1 or lowest of the lowest else crossdev will vomit ebuild links in other non-desired repos)` 
_user overlays if contributed : gpg sign of files might be desired optionally_ 
gpg-key:hash  , sign package manifests :true  
 gpg-sgin tarballs Packages index gen-sha-256 sums :true 
out-hashfile true > 
upload: crossdev-binhost.genpi.com   , else own vps ... etc.   
<config-out: crossdev-overlay.json> 
llvm x86 amd64 , arm64 arm  mips... risc -v nvidia amd .... etc etc use flags .. rust-nightly etc.  clang etc.
gcc-win-x86 , amd64-win-gcc , arm64-win-gcc , arm7hf-gcc-win....  linux-cross-gcc (%targets) 
 
gcc-pass 1   Min use,  ie bootstrap ,  per-wiki 
gcc-pass 4/5 : ztsd go pgo etc etc  hardened , etc etc.   rebuild gcc's with use case flags.   
along with works loads of the crossdev/pentesting community needs  having a useful overlay for switched the default built clang or rust too can be useful... 
ie re-compile firmware for mips??? wifi-cac..... dongles.. 

detect if on native host ie rpi 4 or hopefully soon arm64 server march-ish... '21....   however the ability to spider overlay's 

ie gentoo-haskell  quickpkg all if ::haskell > haskell.genpie64.com / else etc.. 

or packages to overlay repos ie lutris snes9x dolphin ,  ie games.genpie64.com

qt upgrading with vulkan enabled to have some rpi4 support for soon to release drivers

samip5, 2021-02-03:
This is very difficult to understand, and I don't see it being useful at this time.

I'm marking this as wontfix.

closed 2021-02-03 by samip5

necrose99, 2021-02-04:
Yaml can be a touch easier to read , at times. 
Im dyslexic,   so having a yaml file to edit , then sync back to json or parse yaml back to json 

Bit of a creature confort ask.