GenPi64/Build.Dist #14: Running pychroot messes up cgroups on systemd hosts

jonesmz, 2020-12-30:
If we make an ebuild to allow portage to install the project, we'll want to have a USE flag to switch between using pychroot or systemd-nspawnd, to avoid screwing up systemd systems.

samip5, 2020-12-30:
Hmm, interesting. Mine is running systemd but no such issues. Is your host system Gentoo too?

perkinslr, 2020-12-30:
Probably need to abstract the chroot command to a script, since the order of options changes between the two.

jonesmz, 2020-12-30:
> Hmm, interesting. Mine is running systemd but no such issues. Is your host system Gentoo too?

Yes, it is.

It's not clear to me exactly what messed it up, but I have a few hundred thousand journal entries complaining that systemd couldn't manage cgroups, and ssh stopped working :-)

perkinslr, 2020-12-30:
Might the lack of xattr support in qemu be a contributing factor?

jonesmz, 2020-12-30:
I can't say that it isn't possible, but I would be pretty confused how the two would interact like that.

perkinslr, 2020-12-30:
pychroot pulls some cgroup/namespace magic to get sockets, symlinks, and other things from the host environment passed into the chroot in a transparent fashion.  I would not be surprised if some of those sockets had xattrs which qemu then broke.

perkinslr, 2021-03-22:
There is now a wrapper script that uses `CHROOT_CMD` (default `pychroot` or `systemd-nspawn`).  Autodetecting systemd would be good though...

closed 2021-03-30 by jonesmz