GenPi64/Build.Dist #39: Change GenPi64 and GenPi64Desktop to GenPi64OpenRC and GenPi64OpenRC

jonesmz, 2021-01-30:
I'm working on systemd support (for the image, not the host), and have found a few places where i need to override settings from the base profile.

It would be more ergonomic for systemd support if we could have only truly init-system agnostic settings in the Base and GenPi64 configs in config/config.py, with appropriate GenPi64OpenRC and GenPi64DesktopOpenRC profiles inheriting from them.

perkinslr, 2021-01-30:
That sounds like a good idea.  Neat thing with the config system is we can even compose things together (inherit from multiple bases).  What sorts of things are implicitly tied to OpenRC?

jonesmz, 2021-01-30:
I'm not too far down this process yet, so I don't yet have an much of an answer. But here's my addition so far.
```
+GenPi64Systemd = GenPi64 | {
+    "stage3": os.environ.get("STAGE3", "stage3-arm64-systemd.tar.xz"),
+    "stage3url": "http://distfiles.gentoo.org/releases/arm64/autobuilds/latest-stage3-arm64-systemd.txt",
+    "stage3mirror": "http://distfiles.gentoo.org/releases/arm64/autobuilds/",
+    "profile": "genpi64:default/linux/arm64/17.0/genpi64-systemd",
+    "portage": GenPi64["portage"] | {
+        "make.conf": GenPi64["portage"]["make.conf"] | {
+            "USE": GenPi64["portage"]["make.conf"]["USE"] + ["systemd", "-elogind"]
+        }
+    },
+    "overlays": Base['overlays'] + [
+        {
+            'name': 'genpi64',
+            'location': '/var/db/repos/genpi64',
+            'sync-type': 'git',
+            'sync-uri': 'https://github.com/jonesmz/genpi64-overlay.git',
+            'priority': '100',
+            'auto-sync': 'yes',
+            'clone-depth': '1',
+            'sync-depth': '1',
+            'sync-git-clone-extra-opts': '--single-branch --branch systemd'
+        },
+        {
+            'name': 'genpi-tools',
+            'location': '/var/db/repos/sakaki-tools',
+            'sync-type': 'git',
+            'sync-uri': 'https://github.com/GenPi64/sakaki-tools.git',
+            'priority': '50',
+            'auto-sync': 'yes',
+            'clone-depth': '1',
+            'sync-depth': '1',
+        }
+    ]
+}
```

perkinslr, 2021-02-21:
Done.

closed 2021-02-21 by perkinslr