GenPi64/Build.Dist #40: Have the builder take a btrfs snapshot after each stage completes

jonesmz, 2021-01-30:
The reason for this would be to support resuming from an arbitrary point in the build process, even after other steps have finished.

perkinslr, 2021-01-30:
Since we use the native filesystem for building, this would require the host system be using btrfs.  It's a good idea, but should have an environment variable to toggle it on and off (I'm fine defaulting to on, but not picky either way).  Also wouldn't be amiss to have it call a script to do the snapshotting, in case we decide we need to handle other filesystems or snapshot methods later.

jonesmz, 2021-01-30:
>  It's a good idea, but should have an environment variable to toggle it on and off  

Of course

>  Also wouldn't be amiss to have it call a script to do the snapshotting, in case we decide we need to handle other filesystems or snapshot methods later.

Agreed

perkinslr, 2021-02-21:
Use `export BTRFS_SNAPSHOTS=/path/to/hold/snapshots` to enable this.

closed 2021-02-21 by perkinslr