GenPi64/Build.Dist #50: Add nodiratime flag to / partition

hrk, 2021-02-03:
Same rationale as #42

perkinslr, 2021-02-03:
Note that turning on noatime implicitly means that nodiratime is enabled as well.  So I don't think setting this flag explicitly is warranted.  The one related flag we might consider is `relatime`, which updates `atime` only if the last `atime` is earlier than the `mtime` (which has implications for programs that track "new" files).  Even that I think is not worth setting, as anyone using one of those programs is likey to want to customize their mount flags regardless.

hrk, 2021-02-03:
I didn't know that noatime included nodiratime. Time to clean up /etc/fstab on my old boxes then... :-)

closed 2021-02-03 by hrk

perkinslr, 2021-02-03:
No worries.  Check `man mount` for details.

hrk, 2021-02-03:
Well, I've been using it way before 2007 ( https://lwn.net/Articles/244941/ ) and never checked the docs again...