GenPi64/Build.Dist #54: kernel=kernel8-rpi.img line not being set in alpha6 branch

jonesmz, 2021-02-13:
But the uart line is.

Bad dependency settings?

jonesmz, 2021-02-13:
Found the problem.

Spaces being injected before and after the = symbol in the ini file.

https://github.com/GenPi64/Build.Dist/pull/55/files is an attempted fix.

perkinslr, 2021-02-13:
See upstream bug https://github.com/pixelb/crudini/issues/33
The issue isn't with `crudini`, it's actually upstream to `crudini` in `dev-python/iniparse`.  Fixing the whitespace around = is trivial, just reflect into `iniparse` to change the default separator (I'll open a PR for that shortly).  

I'll have to poke a bit more to solve the `DEFAULT` issue, but it should be pretty simple too.

perkinslr, 2021-02-13:
The default section issue is because we are explicitly saying `DEFAULT` as the section, which makes it treat it as a proper section instead of an implicit section.  Replace it with an empty string and that problem is solved.

commit 3dbb2191 (2021-02-14): Don't use explicit DEFAULT section with crudini, use empty string
Closes #54
closed 2021-02-16 by perkinslr