GenPi64/Build.Dist #56: Can't resume build from kernel step, because of missing em-config file

jonesmz, 2021-02-13:
If you take an existing image's workspace, and remove some of the lines from the "completed" file, and then re-run the build, the /em-config file won't be present in the chroot, so the build will fail.

It's not sufficient to check to see if the step is listed in the completed file, as subsequent steps can modify the outcomes from previous steps in ways that invalidate them.

Instead, we need each step to have a configurable check that can determine if the step needs to be re-run.

jonesmz, 2021-02-13:
Apparently removing the "config" line from the completed file also doesn't trigger re-running the config step. It skips straight to the step after the last step in the completed file.

perkinslr, 2021-02-14:
This is because the "completed" system is incredibly dumb.  It handles recursive calls to include build lists by just running them with the same `completed` file.  So when you remove a step from the completed list, and that step is from an included file, the fact that the include step itself is marked completed means nothing inside it gets checked again.

As a workaround, you can safely remove all the steps which use the `includejson` parser, as well as whichever step actually needs to be rerun.  

A proper fix will require storing the completed list in a more sane fashion.  At the least, I think includejson should assign a completed file based on the name, so the `gentoo-arm-base` step would write its completed steps out in `gentoo-arm-base.completed`.  It probably *also* needs to note which parser is used for each step, so we can do some intelligent resume logic.  For example, any step which uses `includejson` should probably run *every* time.  Another possibility is to add an `eachrun` dependency field, so that steps which need to run every time the build gets invoked can easily be marked as such.

perkinslr, 2021-02-21:
I think this can be closed now.  You can resume from an arbitrary point by restoring from a snapshot, and if you remove a step from its completed list, it'll get rerun properly.  It's also possible to manually dispatch particular steps after sourcing `env.sh`.

closed 2021-02-21 by perkinslr