GenPi64/Build.Dist #80: When systemd image is working, use systemd-repart for first-boot partition growing

jonesmz, 2021-03-22:
However, note that systemd-repart doesn't appear to understand the idea of changing the filesystem uuid on first boot, so that'll need to be added separately.

perkinslr, 2021-03-22:
That might be a challenge...  I wasn't planning on using a `systemd`-based initramfs, and I doubt `systemd-repart` works in a `busybox` based one.  If we get dracut working with systemd, that might work though.  In the meantime, I think the advantage of a dead-simple initramfs and the *same* initramfs between the two versions is probably worth not trying to make a systemd initramfs manually.

jonesmz, 2022-01-23:
This is installed and configured to run, but won't work while the systemd image uses msdos partitions.

Revisit when GPT (hybrid ?) is used.

closed 2022-01-23 by jonesmz