GenPi64/Build.Dist #81: unpackstage4 parser is including the contents of /var/cache/binpkgs in the image

jonesmz, 2021-03-22:
This seems unnecessary to include in either the tar.xz file, or the image file, since the binary packages should are for the packages that are already installed on the image itself

perkinslr, 2021-03-22:
It shouldn't...  /var/cache/binpkgs is included dynamitcally from $PROJECT_DIR/packages, bind mounted during pychroot runs.  It should not be mounted when tar runs, so the resulting /var/cache/binpkgs should be empty.

perkinslr, 2021-03-22:
Oops, yeah, it mounts packages before packaging.  I think that's a leftover from when the build system stopped at creating the stage4 tarball.  Including the packages made sense in that case, as you might uninstall and reinstall some package, same reason for including packages which aren't installed back in the days of CD/DVD distros of linux.  Not needed in our case, especially as the clean step should just strip them anyway.

closed 2021-04-28 by perkinslr