GenPi64/Build.Dist #94: Systemd image has no scrollback on conaole.

jonesmz, 2021-03-26:
Shift + page up has no effect.

I'm not sure if this is an issue on the openrc image as well, but I'm making an issue so I don't forget to fix it for the systemd image.

k9spud, 2021-03-26:
The Linux Kernel 5.9+ dropped support for shift-PgUp console scrollback features due to bugs and lack of maintainers/users:

https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.9-Drops-Soft-Scrollback

jonesmz, 2021-03-28:
Please don't close issues that I file. I opened it because I intend to fix it.

Whether i fix it by using an alternative to agetty, or something else.

jonesmz, 2021-03-28:
The kmscon package provides an alternative to getty that includes the shift+pageup feature.
It works in x86_64, but i haven't tested it in aarch64.

samip5, 2021-03-28:
I have an idea, to prevent other people from closing the issues. Could you please self-assign the issues that you intend to do yourself?

perkinslr, 2021-04-13:
`kmscon` and `fbterm` both may do what you want.  Unfortunately, both are kinda dead projects, so bitrot is likely a problem, or will be.  I use `kmscon` on a laptop, and while it's neat, it isn't the most stable.  At a minimum, you'll need to tweak a setting to make backspace behave itself properly.

Have you considered just using `app-misc/screen` or `termux`?