GenPi64/Build.Dist #100: Building gpt partitioned image fails

jonesmz, 2021-03-31:
This configuration chunk in config/config.py

```
    'image': {
        'name': 'GenPi64.img',
        'size': '8G',
        'format': 'gpt',
        'mount-order': [1, 0],
        'uuid': UUID[:8],
        'partitions': [
            {
                'end': '256MiB',
                'format': 'vfat',
                'mount-point': '/boot',
                'mount-options': 'noatime',
                'flags': {
                    'lba': 'on'
                }
            },
            {
                'end': '100%',
                'format': 'btrfs',
                'mount-point': '/',
                'mount-options': 'noatime,compress=zstd:15,ssd,discard',
                'args': f'--force'
            }
]
    }
```

Results in this build output

```
manifest= [{'name': 'makeimage', 'deps': [], 'parser': 'makeimage', 'args': ['']}, {'name': 'mountimage', 'deps': ['makeimage'], 'parser': 'mountimage', 'args': ['']}, {'name': 'unpackstage4', 'deps': ['mountimage'], 'parser': 'unpackstage4', 'args': ['']}, {'name': 'fstab', 'deps': ['unpackstage4'], 'parser': 'fstab', 'args': ['']}, {'name': 'cmdline', 'deps': ['fstab'], 'parser': 'cmdline', 'args': ['']}, {'name': 'firstboot', 'deps': ['cmdline'], 'parser': 'firstboot', 'args': ['']}, {'name': 'strip', 'deps': ['firstboot'], 'parser': 'strip', 'args': ['']}, {'name': 'shrink-image', 'deps': ['strip'], 'parser': 'shrink-image', 'args': ['']}, {'name': 'unmount-image', 'deps': ['shrink-image'], 'parser': 'unmount', 'args': ['']}, {'name': 'compress-image', 'deps': ['unmount-image'], 'parser': 'compress-image', 'args': ['']}]
checking  makeimage deps []
all deps satisfied for makeimage
executing  makeimage deps []
running jobs [(<Popen: returncode: None args: ['/root/build/systemd/parsers/makeimage/makei...>, 'makeimage')]
rm -f GenPi64.img
truncate -s 8G GenPi64.img
parted --script GenPi64.img mklabel gpt
parted --script GenPi64.img mkpart primary 1MiB 256MiB
parted --script GenPi64.img mkpart primary 256MiB 100%
parted --script GenPi64.img print
Model:  (file)
Disk /root/genpi64/systemd/GenPi64.img: 8590MB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags:

Number  Start   End     Size    File system  Name     Flags
 1      1049kB  268MB   267MB                primary
 2      268MB   8589MB  8320MB               primary

sync
losetup -P -f GenPi64.img
mkfs.vfat /dev/loop16p1
mkfs.fat 4.2 (2021-01-31)
parted --script GenPi64.img set 1 lba on
parted: invalid token: lba
FATAL: JOBFAILED  makeimage
FATAL: JOBFAILED  packgenpi
run complete.
```

Looks like the part of the makeimage script that deals with the lba flag isn't working with gpt.

samip5, 2021-03-31:
Did you try to change the lba flag in config to boot?

jonesmz, 2021-04-06:
Switching from "lba" to "boot" resolved this problem.