GenPi64/Build.Dist #122: For unknown reason, when building with systemd-nspawn jobs exit as fatal even though they're not failing

samip5, 2021-11-06:
I don't know why this happens but apperantly this is fatal for no reason?

Building the GenPi64OpenRC target.

```
running jobs [(<Popen: returncode: None args: ['/root/Build.Dist/parsers/rawcommand/rawcomm...>, 'locale'), (<Popen: returncode: None args: ['/root/Build.Dist/parsers/rawcommand/rawcomm...>, 'eixupdate')]
Running eix-update
Spawning container 3d30eef5-243c-4794-b5a3-969c9c4cfd72 on /root/Build.Dist/build/GenPi64OpenRC/chroot.
Press ^] three times within 1s to kill container.
Directory tree /root/Build.Dist/build/GenPi64OpenRC/chroot is currently busy.
Reading Portage settings...
Building database (/var/cache/eix/portage.eix)...
[0] "gentoo" /var/db/repos/gentoo/ (cache: metadata-md5-or-flat)
     Reading category 167|167 (100) Finished             
[1] "genpi-tools" /var/db/repos/genpi-tools (cache: parse|ebuild*3.0.24#metadata-md5#metadata-flat#assign)
     Reading category   1|167 (  0): acct-group...FATAL: JOBFAILED  locale
                                                                          Waiting for jobs to complete...
                                                                                                         running jobs [(<Popen: returncode: None args: ['/root/Build.Dist/parsers/rawcommand/rawcomm...>, 'eixupdate')]
     Reading category 167|167 (100) Finished       
[2] "genpi64" /var/db/repos/genpi64 (cache: parse|ebuild*3.0.24#metadata-md5#metadata-flat#assign)
     Reading category  42|167 ( 25): dev-libs... * ERROR: dev-libs/pigpio-71-r1::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 * 
 * Call stack:
 *                ebuild.sh, line 657:  Called source '/var/db/repos/genpi64/dev-libs/pigpio/pigpio-71-r1.ebuild'
 *      pigpio-71-r1.ebuild, line   8:  Called inherit 'distutils-r1' 'systemd' 'toolchain-funcs'
 *                ebuild.sh, line 337:  Called __qa_source '/var/db/repos/gentoo/eclass/distutils-r1.eclass'
 *                ebuild.sh, line 114:  Called source '/var/db/repos/gentoo/eclass/distutils-r1.eclass'
 *      distutils-r1.eclass, line 105:  Called inherit 'python-r1'
 *                ebuild.sh, line 337:  Called __qa_source '/var/db/repos/gentoo/eclass/python-r1.eclass'
 *                ebuild.sh, line 114:  Called source '/var/db/repos/gentoo/eclass/python-r1.eclass'
 *         python-r1.eclass, line 250:  Called _python_set_globals
 *         python-r1.eclass, line 198:  Called _python_set_impls
 *   python-utils-r1.eclass, line 159:  Called die
 * The specific snippet of code:
 *                      die "No supported implementation in PYTHON_COMPAT."
 * 
 * If you need support, post the output of `emerge --info '=dev-libs/pigpio-71-r1::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=dev-libs/pigpio-71-r1::genpi64'`.
 * Working directory: '/usr/lib/python3.9/site-packages'
 * S: '/pigpio-71'

ebuild failed with status 1
     Reading category  42|167 ( 25): dev-libs...
cannot properly execute /var/db/repos/genpi64/dev-libs/pigpio/pigpio-71-r1.ebuild
     Reading category  49|167 ( 29): dev-qt... * ERROR: dev-qt/qtwebengine-5.15.2::genpi64 failed (depend phase):
 *   qt5-build: EAPI 7 not supported
 * 
 * Call stack:
 *                   ebuild.sh, line 657:  Called source '/var/db/repos/genpi64/dev-qt/qtwebengine/qtwebengine-5.15.2.ebuild'
 *   qtwebengine-5.15.2.ebuild, line   7:  Called inherit 'multiprocessing' 'python-any-r1' 'qt5-build'
 *                   ebuild.sh, line 337:  Called __qa_source '/var/db/repos/gentoo/eclass/qt5-build.eclass'
 *                   ebuild.sh, line 114:  Called source '/var/db/repos/gentoo/eclass/qt5-build.eclass'
 *            qt5-build.eclass, line  20:  Called die
 * The specific snippet of code:
 *      *) die "${ECLASS}: EAPI ${EAPI:-0} not supported" ;;
 * 
 * If you need support, post the output of `emerge --info '=dev-qt/qtwebengine-5.15.2::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=dev-qt/qtwebengine-5.15.2::genpi64'`.
 * Working directory: '/usr/lib/python3.9/site-packages'
 * S: '/qtwebengine-5.15.2'

ebuild failed with status 1
     Reading category  49|167 ( 29): dev-qt...
cannot properly execute /var/db/repos/genpi64/dev-qt/qtwebengine/qtwebengine-5.15.2.ebuild
     Reading category  90|167 ( 53): mail-client... * ERROR: mail-client/thunderbird-60.8.0-r1::genpi64 failed (depend phase):
 *   No supported implementation in PYTHON_COMPAT.
 * 
 * Call stack:
 *                      ebuild.sh, line 657:  Called source '/var/db/repos/genpi64/mail-client/thunderbird/thunderbird-60.8.0-r1.ebuild'
[... 250 lines cut ...]
 *         python-r1.eclass, line 250:  Called _python_set_globals
 *         python-r1.eclass, line 198:  Called _python_set_impls
 *   python-utils-r1.eclass, line 159:  Called die
 * The specific snippet of code:
 *                      die "No supported implementation in PYTHON_COMPAT."
 * 
 * If you need support, post the output of `emerge --info '=media-video/mpv-0.31.0-r1::genpi64'`,
 * the complete build log and the output of `emerge -pqv '=media-video/mpv-0.31.0-r1::genpi64'`.
 * Working directory: '/usr/lib/python3.9/site-packages'
 * S: '/mpv-0.31.0'

ebuild failed with status 1
     Reading category 102|167 ( 61): media-video...
cannot properly execute /var/db/repos/genpi64/media-video/mpv/mpv-0.31.0-r1.ebuild
     Reading category 167|167 (100) Finished        
Applying masks...
Calculating hash tables...
Writing database file /var/cache/eix/portage.eix...
Database contains 19578 packages in 167 categories

Container 3d30eef5-243c-4794-b5a3-969c9c4cfd72 exited successfully.
jobdone eixupdate
FATAL: JOBFAILED  gentoo-base
run complete.
```

samip5, 2021-11-06:
@jonesmz Have you came across this issue? A quite few of the jobs "fail" and it has to be re-started.

jonesmz, 2021-11-06:
No, i have not seen what you're describing.

But aren't these actually failing?

> ebuild failed with status 1
>      Reading category  90|167 ( 53): mail-client...
> cannot properly execute /var/db/repos/genpi64/mail-client/thunderbird/thunderbird-68.6.0.ebuild
>      Reading category [...]

samip5, 2021-11-06:
> No, i have not seen what you're describing.
>
> But aren't these actually failing?

Technically ebuilds are but eix is not.

jonesmz, 2021-11-06:
What's the actual exit code from eix?

samip5, 2021-11-06:
> What's the actual exit code from eix?

How would I know that?

jonesmz, 2021-11-06:
eix is almost certainly exiting with a non-0 exit code, and that's why systemd-nspawn is terminating.

samip5, 2021-11-06:
> eix is almost certainly exiting with a non-0 exit code, and that's why systemd-nspawn is terminating.

No, it would appear to not be systemd-nspawn but rather the python portion itself. Read it fully.

jonesmz, 2021-11-06:
eix-sync returns non-zero, so systemd-nspawn returns non-zero, so build.dist stops