GenPi64/Build.Dist #123: The makeimage parser needs to unregister the loop device on exit.

jonesmz, 2021-12-04:
Failing to do that leaves the loop device setup perpetually if a problem happens.

jonesmz, 2022-05-28:
In a bash script, we would do this as a trap that calls a function on exit.

I'm not sure how to accomplish it in python though.

samip5, 2024-01-13:
We could probably do it eg:

Using atexit module:
```python
import atexit

def cleanup():
    # Detach the loop device associated with the image file
    system(f"losetup --detach {LOOP}")

# Register the cleanup function to be called on exit
atexit.register(cleanup)

# ... (rest of your script)
```
or by using signals:
```python
import signal

def cleanup(signum, frame):
    # Detach the loop device associated with the image file
    system(f"losetup --detach {LOOP}")
    sys.exit(0)

# Register the signal handler for SIGTERM and SIGINT
signal.signal(signal.SIGTERM, cleanup)
signal.signal(signal.SIGINT, cleanup)

# ... (rest of your script)
```