GenPi64/Build.Dist #146: jenkins: Set up git reference repository per agent for each git repository used in a build

jonesmz, 2022-02-06:
The purpose of doing this would be to reduce the amount of git traffic necessary to build.

I've observed github doing fairly aggressive rate limiting in the past for build servers, and caching the repository's chunks in a reference repo on the local disk was very helpful for this.

samip5, 2022-02-06:
Which has own problema from limited storage on the builders.

jonesmz, 2022-02-06:
I suspect that a git repo cache would be so small that it wouldn't really be noticable.

The purpose is just to avoid hitting the git ratelimiting, which deals with transactions in addition to overall bandwidth. It's the transaction limit i'm concerned about.

jonesmz, 2022-07-11:
Implemented

closed 2022-07-11 by jonesmz