GenPi64/Build.Dist #148: Failed build step resulting in build.sh returning success

jonesmz, 2022-02-06:
```
[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1342)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1343)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1344)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1345)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1346)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1347)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1348)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1349)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1350)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1351)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1352)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1353)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1354)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1355)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1356)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1357)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1358)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1359)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1360)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1361)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1362)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1363)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1364)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1365)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1366)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1367)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1368)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1369)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1370)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1371)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1372)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1373)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1374)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1375)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1376)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1377)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1378)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1379)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1380)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1381)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1382)[](https://ci.genpi64.com/blue/organizations/jenkins/GenPi64%2FBuild.Dist/detail/jenkins/17/pipeline/62#step-67-log-1383)['', 'downloadstage3', 'stage3', 'qemu', 'config', 'overlays', 'git', 'portage', 'etc', 'removeoldgcc', 'enablebinpkg', 'rebuildinstalled', 'zapcertifi', 'certifi', 'screen', 'update_portage']

manifest= [{'name': 'locale', 'deps': [], 'parser': 'rawcommand', 'args': ['eselect locale set en_US.utf8']}, {'name': 'users', 'deps': [], 'parser': 'users', 'args': ['']}, {'name': 'eix', 'deps': [], 'parser': 'emerge', 'args': ['eix']}, {'name': 'eixupdate', 'deps': ['eix'], 'parser': 'rawcommand', 'args': ['eix-update']}, {'name': 'bison', 'deps': ['eixupdate'], 'parser': 'emerge', 'args': ['-v1 bison']}, {'name': 'updatelibc', 'deps': ['bison'], 'parser': 'emerge', 'args': ['-v1bg --update --deep  --newuse --jobs glibc']}, {'name': 'emergesystem', 'deps': ['updatelibc'], 'parser': 'emerge', 'args': ['-v1bg --update --deep --newuse --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @system', '-X sqlite -introspection -gtk -elogind -policykit -qt5 -qt4']}, {'name': 'setprofile', 'deps': ['emergesystem'], 'parser': 'setprofile', 'args': ['']}, {'name': 'emergeprofile', 'deps': ['setprofile'], 'parser': 'emerge', 'args': ['-v1bg --update --deep --newuse --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @system']}, {'name': 'addsets', 'deps': ['emergeprofile'], 'parser': 'addsets', 'args': ['']}, {'name': 'setuptools', 'deps': ['addsets'], 'parser': 'emerge', 'args': ['-v1bg --update --newuse dev-python/setuptools']}, {'name': 'emergeworld', 'deps': ['setuptools'], 'parser': 'emerge', 'args': ['-v1bg --deep --update --newuse --with-bdeps=y --keep-going --jobs $(/em-config jobs) --load-average $(/em-config load-average) --tree --autounmask-backtrack=y --backtrack=3000 @world'], 'env': {'SUPPRESS_BASHRC_MAKEOPTS': '1'}}, {'name': 'depclean', 'deps': ['emergeworld'], 'parser': 'emerge', 'args': ['-v --depclean']}, {'name': 'setupservices', 'deps': ['depclean'], 'parser': 'setupservices', 'args': ['']}, {'name': 'news', 'deps': ['setupservices'], 'parser': 'rawcommand', 'args': ['eselect news read &>/dev/null']}, {'name': 'set_root_password', 'deps': ['news'], 'parser': 'password', 'args': ['']}]

checking  locale deps []

all deps satisfied for locale

executing  locale deps []

checking  eix deps []

all deps satisfied for eix

executing  eix deps []

checking  bison deps ['eixupdate']

checking  updatelibc deps ['bison']

checking  emergesystem deps ['updatelibc']

checking  setprofile deps ['emergesystem']

checking  emergeprofile deps ['setprofile']

checking  addsets deps ['emergeprofile']

checking  setuptools deps ['addsets']

checking  emergeworld deps ['setuptools']

checking  depclean deps ['emergeworld']

checking  setupservices deps ['depclean']

checking  news deps ['setupservices']

checking  set_root_password deps ['news']

running jobs [(<Popen: returncode: None args: ['/home/jenkins/workspace/GenPi64_Build.Dist_...>, 'locale'), (<Popen: returncode: None args: ['/home/jenkins/workspace/GenPi64_Build.Dist_...>, 'eix')]

FATAL: JOBFAILED  eix

Waiting for jobs to complete...

running jobs [(<Popen: returncode: None args: ['/home/jenkins/workspace/GenPi64_Build.Dist_...>, 'locale')]

jobdone locale

loading manifest

load complete

Loading status

status not loadable

load complete

[... cut ...] rgs': ['############################\n# Build complete\n# To install your image on an SD card, run\n# dd if=/path/to/image of=/path/to/sdcard\n#############################\n']}]

checking  gentoo-base deps []

all deps satisfied for gentoo-base

executing  gentoo-base deps []

checking  echo-lite-banner deps ['genpi64-lite-finalize']

running jobs [(<Popen: returncode: None args: ['/home/jenkins/workspace/GenPi64_Build.Dist_...>, 'gentoo-base')]

FATAL: JOBFAILED  gentoo-base

run complete.
```

samip5, 2022-02-06:
The eix step often does fail.

jonesmz, 2022-02-06:
yep. Though it shouldn't be resulting in the build continuing and then build.sh returning 0.

i think better to remove the eix-sync from the image entirely. While it's not terrible to have, it's not part of the core gentoo experience and end-users can install it for themselves trivially.

That being said, removal of eix can be it's own ticket.

jonesmz, 2022-07-11:
This might be caused by https://github.com/GenPi64/Build.Dist/issues/166