GenPi64/Build.Dist #161: Use host:/var/db/repos/gentoo as reference repo, if possible

jonesmz, 2022-06-03:
If the host machine has the folder /var/db/repos/gentoo, and that is a git repository, then adding that as a reference repository during the initial clone of the gentoo repository for portage will drastically improve "download" times for that, by allowing for content chunks that are available locally to be fetched locally.

Once the initial download is completed, we do need to remove the reference link, of course.

samip5, 2022-06-03:
Keep in mind, that it will not always be possible due to not running Gentoo. :)

jonesmz, 2022-06-04:
git already has built in "don't use it if it doesn't work" support, as well as "stop using it after the clone finishes" support. All great.

I've implemented this change in the jenkins-aws branch, as well as providing an environment variable to specify a cache-dir to use.

There's a major annoyance though, in that git does not have support for using shallow repositories with this reference repo feature.

As such, even a shallow repo that has all of the commits that we want can't be used to accellerate the network traffic.

In the interrium, i just let the cache download the entire remote repo, but then still check out the in-image repo as shallow.

jonesmz, 2022-07-11:
Implemented

closed 2022-07-11 by jonesmz