GenPi64/Build.Dist #175: Fix build.sh's emerge console output to be a bit friendlier

jonesmz, 2022-12-31:
Specifically the "Load avg: 4.43, 4.95, 4.92" output that puts quite a lot of unnecessary linebreaks in the output.

Upstream link: https://bugs.gentoo.org/889224 partial improvement https://github.com/GenPi64/Build.Dist/pull/174

jlpoolen, 2023-01-02:
Not finished, but the regular expression does what I think is needed --- removing the intermittent status lines that simply have a carriage return and no line-feed.

I've left in my testing stuff so the script is readily adaptable for further quick tests and sample data for a quick run.  
The sample data should have some carriage returns only, an no line feeds, in it.  One has to be careful as clipboards tend to add line feeds to carriage returns.  Add suffix ".txt" since GitHut does not allow Perl scripts to be dropped into this interface.

[remove_statuses - Copy.pl.txt](https://github.com/GenPi64/Build.Dist/files/10329735/remove_statuses.-.Copy.pl.txt)

jlpoolen, 2023-01-02:
Here's a comparison:

Jenkins output downloaded and run through ansi2html: https://salemdata.net/GenPi64/2022_12_30_Build_log_unmodified.txt.html

Output from remove_statuses_C.pl after removing carriage return entries within a line, adding line numbers, and run through ansi2html: https://salemdata.net/GenPi64/output.txt.html

I would like to  have the ability to keep active the line entry when a job is completed as it will allow me to parse the log file and determine the actual amount of time a specific package took to build.