GenPi64/Build.Dist #194: Request To Activate Wiki

jlpoolen, 2023-02-03:
Logan,

Would it be possible to activate the wiki for the Build.Dist project?

I have some interesting data about the builds after parsing the emerge log that I'd like to share and get some input on and a wiki would be an excellent way to collaborate.

jlpoolen, 2023-02-03:
For instance, for run 42 of January 28, the "Lite" process, I have entries relating to the ncurses package:

    emergedb=> select line, to_timestamp(epoch) as "Date",
    emerge_batch_id, code, action
    from emerge_event
    where package_name like '%ncurses%'
    and emerge_batch_id in (
       select id
       from emerge_batch
       where jenkins_cat_id = 1
    )
    order by id;
     line |          Date          | emerge_batch_id | code |       action
    ------+------------------------+-----------------+------+---------------------
       44 | 2023-01-28 22:32:29-08 |               3 | >>>  | start
       85 | 2023-01-28 22:33:10-08 |               3 | ===  | compiling-packaging
      248 | 2023-01-28 22:38:04-08 |               3 | >>>  | autoclean
      249 | 2023-01-28 22:38:04-08 |               3 | ===  | unmerging
      250 | 2023-01-28 22:38:09-08 |               3 | >>>  | unmerge completed
      251 | 2023-01-28 22:38:13-08 |               3 | ===  | Post-Build cleaning
      252 | 2023-01-28 22:38:13-08 |               3 | :::  | completed
     1837 | 2023-01-28 23:39:13-08 |               5 | >>>  | start
     1895 | 2023-01-28 23:39:53-08 |               5 | ===  | compiling-packaging
     2208 | 2023-01-28 23:47:44-08 |               5 | >>>  | autoclean
     2209 | 2023-01-28 23:47:44-08 |               5 | ===  | unmerging
     2210 | 2023-01-28 23:47:48-08 |               5 | >>>  | unmerge completed
     2211 | 2023-01-28 23:47:52-08 |               5 | ===  | Post-Build cleaning
     2212 | 2023-01-28 23:47:52-08 |               5 | :::  | completed
     4441 | 2023-01-29 03:44:49-08 |               6 | >>>  | start
     4499 | 2023-01-29 03:45:30-08 |               6 | ===  | compiling-packaging
     4812 | 2023-01-29 03:52:33-08 |               6 | >>>  | autoclean
     4813 | 2023-01-29 03:52:33-08 |               6 | ===  | unmerging
     4814 | 2023-01-29 03:52:37-08 |               6 | >>>  | unmerge completed
     4815 | 2023-01-29 03:52:40-08 |               6 | ===  | Post-Build cleaning
     4816 | 2023-01-29 03:52:40-08 |               6 | :::  | completed
    (21 rows)

and the 

    emergedb=> --
    -- What are the emerge batchs for the Lite build?
    --
    select id, line
    from emerge_batch
    where jenkins_cat_id = 1;
     id | line
    ----+------
      1 |    2
      2 |   25
      3 |   41
      4 | 1822
      5 | 1833
      6 | 4437
      7 | 7055
      8 | 7060
      9 | 7916
     10 | 7920
     11 | 7926
     12 | 7933
    (12 rows)

    emergedb=>

It would be helpful to solicit the rational for the 12 emerges to help newcomers to the project understand what the order of processing is and why.

(I'll probably digest the entire emerge log so I can output the entire line being referenced.)

samip5, 2023-02-03:
The wiki is active already though? or which one do you refer to?

jlpoolen, 2023-02-03:
I thought github offers wikis in a project and that it appears along the top line:

[image]

Here's an example from another Github project:

[image]

samip5, 2023-02-03:
> I thought github offers wikis in a project and that it appears along the top line:
> 
> [image]
> 
> Here's an example from another Github project:
> 
> [image]

[image]
Yes? :D

jlpoolen, 2023-02-03:
While your screen shows a wiki, mine still does not.  So perhaps there is a wiki, but the permission are such that I cannot access it.
[image]

samip5, 2023-02-04:
> While your screen shows a wiki, mine still does not. So perhaps there is a wiki, but the permission are such that I cannot access it.

I made a one wiki page, so check if you can access it https://github.com/GenPi64/Build.Dist/wiki

jlpoolen, 2023-02-04:
I can see a heading "wiki" and I can visit the page.  But there is no facility for me to edit the page or contribute.  Here is a screen shot at 8:19 AM PST, February 4, 2023.  It appears the only option I have is to download a copy of the wiki using the button at the bottom right.
[image]

For reference, here is a page from the Moonfire-nvr project where I am able to participate in the wiki.  It looks like the configuration of permissions is the cause here for the discrepancy and places this matter back to Logan.

[image]

samip5, 2023-02-05:
Okay and only person that can do something about is is Logan aka @perkinslr.

jlpoolen, 2023-02-05:
Thank you for trying.

jonesmz, 2023-08-10:
@perkinslr

jlpoolen, 2023-08-17:
@perkinslr

maffblaster, 2024-01-26:
The wiki should be editable now by the community. Please let me know if this is not the case.

closed 2024-01-26 by maffblaster