GenPi64/Build.Dist #209: Remove ccache from final image to be released

jlpoolen, 2023-09-13:
ccache is used during the build process.  Yet, the setting for ccaches persists in the final image's /etc/portage/make.conf so when the image is deployed, the user will receive messages such as "Warning: ccache requested...":

    GenPi64 /home/jlpoole/work/distcc # date; time emerge media-libs/tiff
    Wed Sep 13 02:36:20 PM PDT 2023

    These are the packages that would be merged:

    Calculating dependencies  . ...... done!
    Dependency resolution took 11.21 s.

    [ebuild  N     ] media-libs/tiff-4.5.1  USE="cxx zlib -jbig -jpeg -lzma -static-libs -test -verify-sig -webp -zstd"

    >>> Verifying ebuild manifests
    >>> Emerging (1 of 1) media-libs/tiff-4.5.1::gentoo
    >>> Jobs: 0 of 1 complete, 1 running                Load avg: 0.15, 0.03, 0.01Warning: ccache requested but no masquerade dir can be found in /usr/lib*/ccache/bin
    Warning: ccache requested but no masquerade dir can be found in /usr/lib*/ccache/bin
    >>> Installing (1 of 1) media-libs/tiff-4.5.1::gentoo
    >>> Recording media-libs/tiff in "world" favorites file...
    >>> Completed (1 of 1) media-libs/tiff-4.5.1::gentoo
    >>> Jobs: 1 of 1 complete                           Load avg: 1.20, 0.42, 0.15

     * GNU info directory index is up-to-date.

    real    1m42.271s
    user    1m18.386s
    sys     1m3.153s
    GenPi64 /home/jlpoole/work/distcc #

Work-around is to removed "ccache" from FEATURES in /etc/portage/make.conf

samip5, 2024-01-22:
PR's are accepted. ;)

jonesmz, 2024-01-22:
Having a "during the build" /etc/portage/make.conf and a "after the build" /etc/portage/make.conf parser could be pretty useful in general.

Right now we do some hacky things with regards to the MAKE_OPTS to allow better parallelism during the build than the pi supports.